Lite3DP/Lite3DP-Gen-2 #3: platform tilted during print

chri2, 2024-09-25:
Testprinting the keychain the platform tilted even though I already feared I'd fastened the screw to fix it to strong.

Disassambled the printer to reach the screw on the back of the head holding the paltform and took away the plastic washer and replaced it with a slightly thinner one. Did the same for the screw on the left front of the head.

Before changing the washers the two screws were just visible from inside the head holding the platform, but didn't stick into it.

I had the impression the space between magnet holding the platform and screws to fix it could be bigger so that the screws would grab a bit lower on the metal knob of the platform. This would help to hold the platform more secure.

If you'll need pictures - please let me know.

Lite3DP, 2024-10-22:
Solution added to [Troubleshooting guide](https://github.com/Lite3DP/Lite3DP-Gen-2/tree/main/Troubleshooting).

closed 2024-10-22 by Lite3DP

chri2, 2024-10-25:
There's been some questions about the platform not being hold strong enough in the head on the chat.

To get down to the root of the problem I looked very closely at my Lite3DP.

Here's a drawing the is supposed to show a cut through the head with the knob holding the platform inside it and the one thumb screw:

[image]

Closing the screw should fasten the knob inside the head. 

I observed on my printer that the screw pushes the knob towards the edge of the magnet ring (ruled in the drawing) like shown by the red arrows:

[image]

The green line is supposed to show the surface the knob is pushed against. If the screw is fastened tight, it'll push the knob out of the magnet ring sliding over the green edge as shown here:

[image]

This is where the **two other screws** become important. They can deliver the support from two more sides to hold the knob in place.

**But this only works** if they are not too short so that the knob can be pushed out of the magnet ring by any other screw (letting it slide out of the head while gliding over the edge of the magnet as shown above) **and** if they are not to long to not let the knob settle completely inside the magnet ring.

What I did to adjust the three screws:

* screwed all three screws out as far as I could not see the end pointing inside the head anymore
* placed the platforms knob inside the head on the magnet ring (held by the magnet)
* while holding the platform slightly against the head
  * screwed in the screws each a bit going through all screws again and again **using no force at all**
  * while doing so taking care that the platform wouldn't slide any micrometer out of the magnet ring
* after all screws had been fastened **very, very slightly** without any force and the knob of the platform not moving out of the head a single micrometer
  * fastened the thumb screw moderately
  * fastened the other two screws moderately
  * went through all three of them again fastening them a bit more (but with much less force I'd use on the the thumb screw when preparing a print)
* opened the thumb screw all the way
* took the platform out

I believe at this point I had the correct position for the two other screws to hold the platform with maximum force if the thumb screw is closed and **without pushing the platforms knob out of the magnet ring** anymore.

Without moving the two screws (without the thumb wheel) again I applied locking paint on the inside and outside of the head hoping that it would crouch inside the screw thread a bit and fix the screws in place.

Testing afterwards revealed that the platform is fixed very tight inside the head. In my case this has been a big improvement, because even though I had tried to adjust the two other screws before I hadn't found the right position.

Since my first attempt the [troubleshooting guide](https://github.com/Lite3DP/Lite3DP-Gen-2/blob/main/Troubleshooting/Readme.md) has already been updated.

This is how my adjustment looked afterwards:

[image]
[image]
[image]

@Lite3DP if you want to use any of the drawings, photos or explanation please feel free to copy.

chri2, 2024-10-30:
Trying to print [this relief](https://github.com/Lite3DP/Lite3DP-Gen-2/issues/15#issuecomment-2442390235) lying flat on the platform I found that the platform - even I did all the above - still gradually tilted during the print and the prints came out like this:

[image]

On the left three ones you can see that a cross section would look like a parallelogram. This is because on all three prints the platform tilted:

[image]

To get more friction between the head and the platforms knob I coated the knob of the platform thinly with resin and cured that.

This is how I ended up with the nearly perfect fourth print on the right side:

[image]

When printing large objects flat on the platform this might be a hack to get the platform really secured.

On discord in #general @SPI84 wrote: "After few failures with whole plate prints I've switched plate screws to made from diffrent alloy and additionaly tapared them, finnaly succes" - so replacing the screws might be another option. Hopefully @SPI84 will add details here.

chri2, 2024-10-30:
> On discord in #general @SPI84 wrote: "After few failures with whole plate prints I've switched plate screws to made from diffrent alloy and additionaly tapared them, finnaly succes" - so replacing the screws might be another option. Hopefully @SPI84 will add details here.

[image]
[image]
_© SPI84 on 30.10.2024 on #general Discord channel for Lite3DP Gen2_