Lite3DP/Lite3DP-Gen-2 #4: platform got stuck moving up

chri2, 2024-09-25:
when doing the test print of the _keychain_ with the default parameters at some point the platform didn't move up further than some point and the printer made some mechanical noise. This repeated driving the platform up and down and when the print finished I had to carefully help the platform going up.

The print ended without the _Lite_ imprint on the keychain - obviously because the platform didn't loosen from the release film.

I tried again with slightly adjusted parameters:

Bottom lift height: 8.00 mm
Lift height: 6.00 mm
Lift speed: 1.30 mm/s

With those changes it worked.

andypiper, 2024-09-26:
what / where are the keychain and touchpen test prints you refer to in this issue and other issues?

Lite3DP, 2024-10-22:
Ensure proper platform adjustment and adequate lifting speed.

closed 2024-10-22 by Lite3DP