Lite3DP/Lite3DP-Gen-2 #5: The progression display is printed

neolao, 2024-09-25:
I tried to print the touchpen and the result is this:

[image]

You can see "99" and "TO FINISH". 
I suppose that the image destined to the touch screen was used to print.
I can also say that the images (the sliced model + touch screen display) are merged.

chri2, 2024-09-25:
I had a similar issue: I selected a folder from the sdcard and briefly a yellow printed `Preview` appeared at the bottom of a otherwise black screen. Then the screen jumped to the `PARAMETERS` menu page.

I started the print and got something I think is the menu.
[image]

jene8774, 2024-09-26:
+1 
It seems to be a bug that happens across the board, not an individual error.

chri2, 2024-09-29:
To check @jene8774 idea about the additional files which are not images of layers I started a print as generated by PrusaSlicer. The directory contains the following files:
```
$ ls -nl | head -n4; echo '[...]'; ls -nl | tail -n4
insgesamt 2336
-rw-r--r-- 1 1000 1000  419 29. Sep 13:31 config.ini
-rw-r--r-- 1 1000 1000 3004 29. Sep 13:31 prusaslicer.ini
-rw-r--r-- 1 1000 1000 2500 29. Sep 13:31 rmote300000.png
[...]
-rw-r--r-- 1 1000 1000  333 29. Sep 13:31 rmote300578.png
-rw-r--r-- 1 1000 1000  318 29. Sep 13:31 rmote300579.png
-rw-r--r-- 1 1000 1000  287 29. Sep 13:31 rmote300580.png
drwxr-xr-x 2 1000 1000 4096 29. Sep 13:32 thumbnail
```

So there are at least two files that are not layer images and maybe the subdirectory will be counted as an additional file also.

neolao, 2024-09-29:
Oh I see. I will try removing unnecessary files

chri2, 2024-09-30:
related to #8

closed 2024-10-22 by Lite3DP