Lite3DP/Lite3DP-Gen-2 #14: tools / clean resin vat - exposure not on whole screen

chri2, 2024-10-22:
I tried the function and here is a picture of what I had:

[image]

I would have expected that I'd get a square without holes

closed 2024-10-22 by Lite3DP

chri2, 2024-10-22:
Exposure time has been 45 seconds. Here are pictures of the UV running without vat and of the screen with lights off:

[image]

[image]

Lite3DP, 2024-10-22:
@chri2 Yes, please check that there is nothing under the display.
Also remember that both the mask display and the touchscreen show the same thing, just that due to the lack of backlight nothing is visible on the mask display. Carefully, with the machine disassembled (without cover or base), you can use a flashlight to illuminate the mask display while you navigate the menu and see better where those spots are coming from. In a very rare case, they may be dead areas on the display, hopefully this is not the case.

chri2, 2024-10-23:
O.k. I took the printer apart and took some pictures. I used some led lights to put lights to the back of the screen while taking the pictures.

## screen off

In these pictures the screen is off and the printer is not powered.

[image]
[image]
[image]
[image]
[image]
[image]
[image]

## screen active

I powered the printer and took these pictures.

[image]

The following pictures are taken while the clean vat tool has been active.

[image]
[image]
[image]

## screen inspection from the back

I carefully controlled the back of the screen for dirt or any signs that may explain the problem seen in the pictures above, but didn't find any. Looking at the silvery back of the screen in daylight the errors from the pictures above can barely be seen as shadows - nearly not notable. I think the visual effect is so weak I can't put it into a photo.

[image]
[image]

Please let me know if I can provide more information to help solve the problem.

Lite3DP, 2024-10-23:
Sorry @chri2 they look like two dead zones on the LCD. Could it be that the edge of the platform hit the display during leveling? I haven't seen anything like this before. It's not the best, but you can still use the machine preventing print sections from including dead zones. However, you might want to order a replacement LCD display from CS before they run out. Sorry again.

chri2, 2024-10-24:
> Sorry @chri2 they look like two dead zones on the LCD. Could it be that the edge of the platform hit the display during leveling? 

No. And it doesn't look like cracks neither like hits.

> I haven't seen anything like this before. It's not the best, but you can still use the machine preventing print sections from including dead zones. However, you might want to order a replacement LCD display from CS before they run out. Sorry again.

Uhm, I'd say this is a production error. I do not see any way how one could inflict a damage like that - like you probably do neither. I'd expect that I'd get provided a screen as a replacement given that I'm able to change it (which probably is possible since there are screens to order).

I do not see where my fault should be in this that I'd have to pay for the replacement screen?

Lite3DP, 2024-10-24:
Customer support contact page: https://www.crowdsupply.com/contact

chri2, 2024-10-25:
Thanks for the suggestion! I did that and am waiting for their answer.

chri2, 2024-10-28:
@Lite3DP, could you provide some advice/guidance on how to remove the screen from the mainboard of the Lite3DP Gen2? Is it glued to the mainboard? Does heat help to free it?

chri2, 2024-10-29:
When I exchanged the screen I found that it is easily peeled off the pcb. It seems only to be held in place by adhesion.

Before removing the screen I opened the [THD THD0515-45CL-SN flat cable connector](https://www.lcsc.com/product-detail/FFC-FPC-Flat-Flexible-Connectors_THD-THD0515-45CL-SN_C283158.html) by carefully pulling up the flap of the [Front Flip-Lock Actuator](https://www.te.com/en/products/connectors/pcb-connectors/wire-to-board-connectors/ffc-fpc-ribbon-connectors/fpc-connectors/intro-to-fine-pitch-fpc-connectors.html) with my fingernail.

The new screen I carefully moved in its place maneuvering its flat cable into the **zero insertion force** (ZIF) connector that I opened to remove the old screen.

My new screen had on its back (silver side) *two* foils: 
* one is large and has a corner to grab to peel it off which I did
* the other one (I guess it is supposed to stick on the bigger first one and get peeled off when pulling the thicker one off, but that didn't work for me) I very carefully lifted with a plastic card at one corner until I could grab it with my fingers.

The upper side (black) had a foil installed also.

My new screen works and has no broken areas.