Lite3DP/Lite3DP-Gen-2 #16: Firmware upload instructions improvements

chri2, 2024-10-30:
Just went through the procedure - **thanks for documenting it so well!**

Here are some additions and corrections that might help to make it even better:

* Add "Please install ESP32 boards core version 2.0.17" from readme.md to the document and add instructions:
  * Add menu location for doing this (Arduino IDE V2.3.3): Tools / Board / Boards Manager
  * Fill _esp32_ into _Filter your search..._
  * In section for _esp32 by Espressif Systems_ use the drop-down list beside the _UPDATE_ or _INSTALL_ button to select Version _2.0.17_
  * hit the _INSTALL_ button
* Add information (URL) about where to get the [Arduino IDE](https://www.arduino.cc/en/software)
* Add menu location for choosing the right board (Arduino IDE V2.3.3): Tools / Board / esp32 / ESP32 Dev Module
* In point 6 add information about whether the printer is allowed to be connected to a 12V power supply while powering it from usb-c via the CP2102 board (I disconnected the main 12V power while flashing)
* In point 9 change _Hit Upload until the upload is complete._ into _Hit Upload and wait until the upload completed._ (the button does not need to be pressed all the time while uploading `*eg*` 😉)

chri2, 2024-10-30:
[image]

ajwatt, 2024-11-14:
I will add an additional clarification for point 7:

- Press and hold the BOOT button, then press and release the RST button, then release the BOOT button.

If you release both buttons at the same time, as the current wording suggests, the printer will simply reboot.